MaJerle/stm32-usart-uart-dma-rx-tx #3: ESP8266 - Examples?

Korporal, 2019-02-23:
Hi,

I was wondering if you have worked with the STM32F769 Discovery board and the dedicated connector it has for an ESP8266?

I want to leverage the async IO pattern that you talk about here but not really sure how much of your source code is portable to the STM32F769, I'm using Visual GDB and Visual Studio for this.

I've actually designed and developed a pretty powerful API in C# that runs on a PC and controls an ESP8266, this is fully async in nature and includes a very flexible and quite powerful RingBuffer class.

I'd now like to rewrite this in C++ and get the board working with the ESP8266.

I can't even find out which USART is tied to the ESP8266 connector, pretty much no documentation.

Thanks

MaJerle, 2019-02-23:
You see the approach, now for the discovery board, you need to check which UART is connected there.
From the link you gave, it is very clear that UART5 is used on this connector, thus you have to:

- Find which TX and RX pins is used for UART5 on the board (PC12 and PD2)
- Open reference manual of STM32F769 MCU and check DMA configuration for UART5 (DMAx, Streamy, Channelz) and use these in example.

Driver for this discovery is available in my ESP library for microcontrollers: https://github.com/MaJerle/ESP_AT_Lib/blob/master/src/system/esp_ll_stm32f769i_discovery.c

closed 2019-02-23 by MaJerle

Korporal, 2019-02-23:
@MaJerle - Hi, and yes you are quite right there is indeed documentation and I was speed reading and doing a poor job! 

Many thanks.

Hugh