MaJerle/stm32-usart-uart-dma-rx-tx #4: Good Idea to add a int16_t uart_get_byte(void)

mgiaco, 2019-04-04:
Hi,

Many thanks for the great samples. Do you have any good idea to add a uart_get_byte? Do you think it is a good idea to add a ringbuffer and then fill this buffer in the usart_rx_check function then I can add a general usart_get_byte() as normal.

thx
mathias

MaJerle, 2019-04-04:
You can use generic ringbuffer and then use `ringbuff_read` to read next character.

I have one: https://github.com/MaJerle/ringbuff

mgiaco, 2019-04-29:
Hello, 
Many thanks. So do you mean that way. 

```c
/**
 * \brief Check for new data received with DMA
 */
void usart_rx_check(void)
{
	static size_t old_pos;
	size_t pos;

	/* Calculate current position in buffer */
	pos = ARRAY_LEN(usart_rx_dma_buffer) - LL_DMA_GetDataLength(DMA1, LL_DMA_CHANNEL_5);
	if (pos != old_pos)
	{ /* Check change in received data */
		if (pos > old_pos)
		{ /* Current position is over previous one */
			/* We are in "linear" mode */
			/* Process data directly by subtracting "pointers" */
//			usart_process_data(&usart_rx_dma_buffer[old_pos], pos - old_pos);
			ringbuff_write(&usart_rx_dma_ringbuff, &usart_rx_dma_buffer[old_pos], pos - old_pos);

		}
		else
		{
			/* We are in "overflow" mode */
			/* First process data to the end of buffer */
//			usart_process_data(&usart_rx_dma_buffer[old_pos], ARRAY_LEN(usart_rx_dma_buffer) - old_pos);
			ringbuff_write(&usart_rx_dma_ringbuff, &usart_rx_dma_buffer[old_pos], ARRAY_LEN(usart_rx_dma_buffer) - old_pos);

			/* Check and continue with beginning of buffer */
			if (pos)
			{
				ringbuff_write(&usart_rx_dma_ringbuff, &usart_rx_dma_buffer[0], pos);
//				usart_process_data(&usart_rx_dma_buffer[0], pos);
			}
		}
	}
	old_pos = pos; /* Save current position as old */

	/* Check and manually update if we reached end of buffer */
	if (old_pos == ARRAY_LEN(usart_rx_dma_buffer))
	{
		old_pos = 0;
	}
}

int16_t usart_get_byte(void)
{
	uint32_t old_primask;
	uint8_t ch;
	uint8_t ret;
	
	/* Check if DMA is active */
	/* Must be set to 0 */
	old_primask = __get_PRIMASK();
	__disable_irq();
	ret = ringbuff_read(&usart_rx_dma_ringbuff, &ch, 1);
	__set_PRIMASK(old_primask);

	if (ret == 0)
	{
		return -1;
	}
	return ch;
}
```

Do you mean that way?
I works but is not so memory efficient because I have added an other ringbuff an now I copy the data between them.

MaJerle, 2019-04-29:
I meant it this way, correct. However, please be advised that if you write to ring buffer from one entry point (in your case you do, you write only from RX thread) and if you read only from one exit point, then you do not need to use interrupt disable feature. Ring buffer library is designed to sustain this approach.

That is less memory efficient, correct, but I believe the right way to do it.

mgiaco, 2019-04-29:
Okay many thanks.

closed 2019-04-29 by MaJerle