MaJerle/stm32-usart-uart-dma-rx-tx #5: help with RX

mjphillips1981, 2019-11-20:
Hello,

First off thank you for these examples.  I am trying to use the usart_rx_idle_line_irq_ringbuff_G0 example.  But would like some guidance on checking what is coming in from the terminal.  I want to send commands from the terminal and act on them but only act on them once.  I have looked in the usart_rx_dma_buffer for my commands and see them, but is this the correct way of doing this?  I want to check some buffer to see if the command is there, act on it, and then clear the buffer.  Any help would be appreciated. 

Thanks,
Mike

MaJerle, 2019-11-20:
If you see received data in the buffer is already a good step further.

Your *processing* function today looks like the one below and does 2 things:
- Writes received data to buffer
- Sends back the same data from the buffer using DMA TX

https://github.com/MaJerle/STM32_USART_DMA/blob/b291d2184dfdd1530028da4b99887005ed738baf/projects/usart_rx_idle_line_irq_ringbuff_G0/Src/main.c#L161-L168

You need to remove function call to start new data transfer, so that finally it only writes data to buffer.

Later, in the main while loop function, you need to check for any data available in the buffer, something like

```c
int
main(void) {
//...
//...
    while (1) {
        if (ringbuff_get_full(&usart_tx_dma_ringbuff) >0) {
            //Buffer has some data, read it using ringbuff_read function
            //and process accordingly
        }
    }
}
```

Are you looking for an example how to do packet communication using DMA RX and later TX?

mjphillips1981, 2019-11-20:
thank you.

Basically what I am trying to do is have a pc application running that someone can command my little control board with.  But I do want the control board to send messages back to the PC as well.

MaJerle, 2019-11-20:
Best is then to have 2 ringbuffers. One used to write rx received data, like now. Second to write tx data and then dma transmits it.

mjphillips1981, 2019-11-20:
i will try that. thank you for the quick response!!

MaJerle, 2019-11-25:
A new project has been added to show:
- How to use 2 ringbuffers, one for RX and one for TX
- How to process RX ringbuff for basic packet control and send it back, all via DMA, no blocking calls.

Examples has not been tested on target board but should work anyway.

closed 2019-12-16 by MaJerle