MaJerle/stm32-usart-uart-dma-rx-tx #6: re-entrance protection?

FrankBau, 2020-01-13:
Thank you very much for your clear and very well documented code examples. I'm using it for F0.

In your code, the interrupt handlers call usart_rx_check which in turn calls usart_process_data. There is a chance that the IDLE interrupt follows very quickly after TC or HT, regardless of the buffer length. 

This can, in theory, cause re-entrancy issues in usart_rx_check() like if the second interrupt falls between reading pos and updating last_pos.

You clear the interrupt flags before the usart_rx_check() call. Wouldn't it be more protective against re-entrancy to clear them only after usart_rx_check() returns? 

Do you see a need for further measures like pausing the DMA etc.?

MaJerle, 2020-01-13:
@FrankBau not sure I understand the issue you are describing.
DMA and UART IDLE have to have same interrupt priority (at least same preemption level) and it will work without problem.

If they do not have same priority levels, then one interrupt could preempt another and this would mean wrong processing and possible corruption of `old_pos` variable.

FrankBau, 2020-01-13:
Oh, I see! Thx for the explanation.

closed 2020-01-13 by FrankBau