MaJerle/stm32-usart-uart-dma-rx-tx #10: /bin/sh: arm-none-eabi-objcopy.exe: command not found

LingxiaoGao, 2020-09-22:
arm-none-eabi-objcopy.exe -O ihex "usart_dma_rx_idle_line_irq_rtos_L4.elf" "usart_dma_rx_idle_line_irq_rtos_L4.hex"
/bin/sh: arm-none-eabi-objcopy.exe: command not found
make[1]: *** [makefile:80: post-build] Error 127
make: *** [makefile:48: all] Error 2
"make -j7 all" terminated with exit code 2. Build might be incomplete.

MaJerle, 2020-09-22:
Did you run it from CubeIDE? If not, you have to setup toolchain on your side.

LingxiaoGao, 2020-09-22:
> to

I am using lasted version of STM32CubeIDE

francescodg4, 2020-10-17:
Quick fix: right click on your project > Properties > C/C++ Build > Settings > Build Steps then remove the command in the "Pre-build steps" tab. It works now on my Linux machine.

Very nice project @MaJerle, keep it up!

closed 2020-11-04 by MaJerle