MaJerle/stm32-usart-uart-dma-rx-tx #15: add "usart_rx_idle_line_irq_ringbuff_rtos" routine

AnthonyYMH, 2021-05-23:
Could you add "usart_rx_idle_line_irq_ringbuff_rtos" routine, thank you.

MaJerle, 2021-05-23:
Combining rtos and ringbuff example should give you expected result.

AnthonyYMH, 2021-05-24:
I have an idea:
1、In the DMA interrupt, write data to the ringbuff.
2、In the UART idle interrupt, also write data to the ringbuff,and send the MessageQueue to the thread,read the ringbuff for data packet processing.
Can I do it like this?
And I don't understand the effect of the HT interrupt.I think it's okay to delete.
Thank you！

MaJerle, 2021-05-24:
No.

In the DMA TC & HT and in USART IDLE irq, do the same thing.
* Write data to ringbuff
* Send message queue to thread

In thread, read ringbuff data and do you stuff.

AnthonyYMH, 2021-05-24:
> No.
> 
> In the DMA TC & HT and in USART IDLE irq, do the same thing.
> 
> * Write data to ringbuff
> * Send message queue to thread
> 
> In thread, read ringbuff data and do you stuff.
> 
> HT irq is indeed very necessary to have stable communication.

Got it, thank you very much for your answer.

closed 2021-05-24 by MaJerle