MaJerle/stm32-usart-uart-dma-rx-tx #21: Using CubeMX and HAL for UART DMA on H7

egoltzman, 2022-05-18:
I'm trying to do the same functionality on H750 but with using CubeMX (From within CubeIDE) and using HAL function (Mainly HAL_UARTEx_ReceiveToIdle_DMA) but it is not working for me.
Is there any reason for you to use the LL over the HAL?

MaJerle, 2022-05-18:
I am using LL drivers to show to full concept to the community - this is main reason actually.

If you want to use it with HAL and CubeMX, ensure that in GUI you have:
- UART is enabled
- UART interrupts are enabled
- UART has configured DMA for RX & DMA has configuration in circular mode

You can use this project from G4 as an idea how to implement it.
https://github.com/STMicroelectronics/STM32CubeG4/blob/master/Projects/NUCLEO-G474RE/Examples/UART/UART_ReceptionToIdle_CircularDMA/Src/main.c

egoltzman, 2022-05-19:
Thank you Tilan for your quick response.
The suggestion to use the G4 example was great and indeed I was able to
port it to my target MCU and run it on the H750 DK successfully.
But, When I tried to use it together with my TouchGFX application (As I
need on my final project target board) the same test that worked before I
added TGFX is not working fine.
I thought it might be FreeRTOS that caused the problem but after adding it
to the first working test it is still working so it looks like it is not
FreeRTOS but TGFX and all its related stuff like DMA2D, LTDC etc. that make
the change.
What I see is that the HAL_UARTEx_RxEventCallback is being called with the
right value in "Size" but the content of the received circular buffer
(aRXBufferUser) holds only the last character that was sent in the first
entry of the buffer., that is, if I send "1234567", I get Size of 7 and the
buffer looks like this:
[image: image.png]
It looks to me like the increment action on the memory somehow is not
working.
What do you think?
Thanks,
Eyal

*Eyal Goltzman               איל גולצמן*
Managing Partner

EmbeddedPartners.co.il <http://embeddedpartners.co.il/>
***@***.***
+972-54-4875-625

egoltzman, 2022-05-19:
Hi Tilen,
This is how the image looks like:
aRXBufferUser[0] is 0x37
aRXBufferUser[1..19] is 0x0

As for the priorities, When I added FreeRTOS the CubeMX updated the UART and DMA priorities to 5 but, as I mention, on the sample test I run without TouchGFX and with FreeRTOS (And the updated priorities of 5) and the UART is still working there as expected.
The problem start when TGFX is added to the project with many other elements (DMA2D. LTDC etc.)

egoltzman, 2022-05-19:
Tilen, I think I know now the reason for the problem, the CubeMX call the initializations of the elments in the wrong order and I need to change it manually in CubeMX to fit this order:
 MX_GPIO_Init();
 MX_CRC_Init();
 MX_LTDC_Init();
 MX_DMA2D_Init();
 MX_FMC_Init();
 MX_MDMA_Init();
 MX_JPEG_Init();
 MX_TouchGFX_Init();
 /* USER CODE BEGIN 2 */
 MX_DMA_Init();
 MX_USART3_UART_Init();

Somehow, someone is braking the DMA settings.
Thanks!
Eyal

closed 2023-04-12 by MaJerle