MaJerle/stm32-usart-uart-dma-rx-tx #22: USART3 Not Found on F411 Nucleo Board

djsg, 2022-06-20:
Hi MaJerle,

I want to study your code, but I only have F411 Nucleo board. 
I found your code for F4 uses UART3 with PD8,9 and Tx/Rx pins. They are not available on my board. Yet, I want to try your code. Any suggestion?

MaJerle, 2022-06-20:
- Select your UART and corresponding pins for it. (see Datasheet or CubeMX)
- Find DMA stream/channels that match your UART configuration (for RX and/or TX) (see reference manual)
- Enable UART & GPIO & DMA clocks
- Configure GPIOs with AF for specific UART (see datasheet)
- Change settings for UART2
- Check Interrupt handler names and vectors
- Check correct flags in DMA/UART interrupt routines.

That would be shortly.

djsg, 2022-06-21:
that readme is very informative. Thx.

closed 2022-06-23 by MaJerle