MaJerle/stm32-usart-uart-dma-rx-tx #29: problem in using multiple DMAs for RX

ymkim92, 2023-04-12:
I love this repo. It has nice documentation as well. Thank you.

I found a problem when I try to work with 2 UARTs by following the instruction here.
The code uses RX DMA. It works fine when I test each UART port one by one. 
But the problem happens when I play with 2 ports at the same time:
- I am sending data to each port every 3 seconds and STM32 echoes it
- It doesn't need to be synchronised. I can see the problem even it is not synchronised.

The problem I found is that "LL_DMA_GetDataLength()" returns a wrong value sometimes very often in the case above.

MaJerle, 2023-04-12:
Please double check that all parameters are properly used for both uarts.

ymkim92, 2023-04-12:
It is very hard to find the cause of the problem. I have decided to use the interrupt instead of DMA.

MaJerle, 2023-04-12:
Can you paste the full source of all uart rx callbacl functions here?

ymkim92, 2023-05-18:
I am sorry for this late response. I can't find the code at the moment. But I guess the cause of the problem was the different priority between interrupt and dma stream. I overlooked the note in README.md.

closed 2023-06-06 by MaJerle