MaJerle/stm32-usart-uart-dma-rx-tx #41: Wrong CubeMX file in 'usart_rx_idle_line_irq_ringbuff_tx_loopback_G4' project

SalmanKhaja, 2024-10-03:
Hi,

The [usart_rx_idle_line_irq_ringbuff_tx_loopback_G4](https://github.com/MaJerle/stm32-usart-uart-dma-rx-tx/tree/main/projects/usart_rx_idle_line_irq_ringbuff_tx_loopback_G4) project have wrong CubeMX (.ioc) file. Please look into it.

MaJerle, 2024-10-03:
By default, this repository is meant to show the concept of DMA + UART implementation down to almost register level with all interrupts.

CubeMX file was never meant to be used as such. I'll. likely delete it.

What in particular is wrong in the .ioc file?

SalmanKhaja, 2024-10-04:
Thanks for clarification, initially I confused that CubeMX is initializing DMA and UART and STM32CubeIDE is throwing error due to mismatch in '.ioc' file. After looking deep into, I have understood that both are initialized on register level in code.

closed 2024-10-04 by SalmanKhaja